abhishek-kakkar/BeagleLogic #7: PRU1 fails to load with bone70 OS

bcr, 2015-06-01:
I think I've followed all the instructions, HDMI is disabled in /boot/uEnv.txt, but PRU1 is failing to load for some reason. Following is my dmesg output.

Any suggestions welcome. If I should take this to a forum, please let me know where to go.

```
[  104.300848] bone-capemgr bone_capemgr.9: part_number 'BB-BEAGLELOGIC', version 'N/A'
[  104.301028] bone-capemgr bone_capemgr.9: slot #7: generic override
[  104.301075] bone-capemgr bone_capemgr.9: bone: Using override eeprom data at slot 7
[  104.301123] bone-capemgr bone_capemgr.9: slot #7: 'Override Board Name,00A0,Override Manuf,BB-BEAGLELOGIC'
[  104.301368] bone-capemgr bone_capemgr.9: slot #7: Requesting part number/version based 'BB-BEAGLELOGIC-00A0.dtbo
[  104.301420] bone-capemgr bone_capemgr.9: slot #7: Requesting firmware 'BB-BEAGLELOGIC-00A0.dtbo' for board-name 'Override Board Name', version '00A0'
[  104.301491] bone-capemgr bone_capemgr.9: slot #7: dtbo 'BB-BEAGLELOGIC-00A0.dtbo' loaded; converting to live tree
[  104.305507] bone-capemgr bone_capemgr.9: slot #7: #2 overlays
[  104.325248] omap_hwmod: pruss: failed to hardreset
[  104.332527] pru-rproc 4a300000.prurproc: #8 PRU interrupts registered
[  104.332705] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [17] <- 1
[  104.332740] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [18] <- 0
[  104.332774] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [19] <- 2
[  104.332806] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [20] <- 3
[  104.332838] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [21] <- 0
[  104.332869] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [22] <- 1
[  104.332901] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [24] <- 4
[  104.332933] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [25] <- 5
[  104.332964] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [26] <- 6
[  104.332996] pru-rproc 4a300000.prurproc: sysevent-to-channel-map [27] <- 7
[  104.333043] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [0] <- 0
[  104.333076] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [1] <- 1
[  104.333107] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [2] <- 2
[  104.333139] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [3] <- 3
[  104.333170] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [4] <- 0
[  104.333201] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [5] <- 6
[  104.333233] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [6] <- 1
[  104.333264] pru-rproc 4a300000.prurproc: channel-to-host-interrupt-map [7] <- 7
[  104.333334] pru-rproc 4a300000.prurproc: found #2 PRUs
[  104.337536] pru-rproc 4a300000.prurproc: No resource-table node node; slave PRU
[  104.337657] pru-rproc 4a300000.prurproc: No resource-table node node; slave PRU
[  104.338068]  remoteproc0: 4a300000.prurproc is available
[  104.338097]  remoteproc0: Note: remoteproc is still under development and considered experimental.
[  104.338122]  remoteproc0: THE BINARY FORMAT IS NOT YET FINALIZED, and backward compatibility isn't yet guaranteed.
[  104.344963]  remoteproc0: powering up 4a300000.prurproc
[  104.401493]  remoteproc0: failed to load beaglelogic-pru1
[  104.407516]  remoteproc0: request_firmware failed: -2
[  104.413016] pru-rproc 4a300000.prurproc: rproc_boot failed
[  104.418950]  remoteproc1: releasing 4a300000.prurproc
[  104.431414] pru-rproc: probe of 4a300000.prurproc failed with error -2
[  104.432783] BeagleLogic loaded and initializing
[  104.432850] beaglelogic: probe of pru-beaglelogic.12 failed with error -1
[  104.432916] bone-capemgr bone_capemgr.9: slot #7: Applied #2 overlays.
root@beaglebone:~# uname -a
Linux beaglebone 3.8.13-bone70 #1 SMP Fri Jan 23 02:15:42 UTC 2015 armv7l GNU/Linux
root@beaglebone:~# cat /etc/dogtag 
BeagleBoard.org Debian Image 2015-03-01
root@beaglebone:~# cat /sys/devices/bone_capemgr.*/slots
 0: 54:PF--- 
 1: 55:PF--- 
 2: 56:PF--- 
 3: 57:PF--- 
 4: ff:P-O-L Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
 5: ff:P-O-- Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI
 6: ff:P-O-- Bone-Black-HDMIN,00A0,Texas Instrument,BB-BONELT-HDMIN
 7: ff:P-O-L Override Board Name,00A0,Override Manuf,BB-BEAGLELOGIC
```

abhishek-kakkar, 2015-06-01:
The firmware is not being loaded it seems.

Do 'beaglelogic-pru0' and 'beaglelogic-pru1' firmware binary files exist in /lib/firmware ? If not, download and copy them into /lib/firmware and then retry.

bcr, 2015-06-01:
Thanks, it looks like there are no beaglelogic files at all in the /firmware directory.
```
root@beaglebone:~# ls /lib/firmware/beaglelogic*
ls: cannot access /lib/firmware/beaglelogic*: No such file or directory
```
Do you have a pointer to where the binaries can be downloaded? I am looking now, I can see the Github source, but I don't know where the binaries are.

abhishek-kakkar, 2015-06-01:
1. https://github.com/abhishek-kakkar/BeagleLogic/releases
2. https://github.com/abhishek-kakkar/BeagleLogic/wiki/Build-BeagleLogic, the link on the top of the page.

bcr, 2015-06-01:
Thanks so much -- sorry I missed the link for the binaries. I think I'm all good now.

FYI, it looks like the Beaglebone distribution isn't including these files right now, and they probably should (the kernel module is there, but as you know these files are required also.)

```
[   85.011097] misc beaglelogic: BeagleLogic PRU Firmware version: 0.2
[   85.013597] misc beaglelogic: Device supports max 128 vector transfers
[   85.013688] misc beaglelogic: Valid PRU capture context structure found at offset 0000
[   85.013753] misc beaglelogic: Default sample rate=50000000 Hz, sampleunit=1, triggerflags=0. Buffer in units of 4194304 bytes each
```

abhishek-kakkar, 2015-06-01:
According to [this link](https://groups.google.com/d/msg/beagleboard/ui8-no8J6HY/JC0hJtL_a8gJ), the files are included in the system image but not in /lib/firmware.

I'll add a note in the documentation regarding this.

Closes #7.

closed 2015-06-01 by abhishek-kakkar

bcr, 2015-06-01:
I commented on that thread, let's see if we can get the files in the right place and not have to futz around moving them if possible.