abhishek-kakkar/BeagleLogic #10: Slow memory access on CPU side?

credentiality, 2016-01-10:
I'm using prussdrv to build a project similar to yours, clocking data into shared DDR with the PRUs.  I get awesome (600MB/sec+) performance writing into shared DDR with PRU0, but when I read it on the linux side, I only get ~30MB/sec tops.  If I switch to another malloc()ed buffer instead of the shared DDR, I get hundreds of MB/sec as expected.

Any idea why I'm only getting 30MB/s reading from the shared memory segment?  Switching between p = local_buf and p = shared_ddr changes performance by >10x:

  prussdrv_map_extmem((void**)&shared_ddr);
  unsigned int shared_ddr_len = prussdrv_extmem_size();
  unsigned int physical_address = prussdrv_get_phys_addr((void*)shared_ddr);

  char *local_buf = (char *) malloc(shared_ddr_len);
...
  while (bCont) {
//    volatile char *p = local_buf;
    volatile char *p = shared_ddr;

    char *p_end = ((uint8_t *)p) + shared_ddr_len;

    volatile char bytes[4096] = {0};
    while (p < p_end) {
      memcpy(bytes, p, sizeof(bytes));

      for (int i = 0; i < sizeof(bytes); i++) {
        sum += bytes[i];
      }

      p += sizeof(bytes);
    }
    local_bufs_written++;
    ...

abhishek-kakkar, 2016-01-10:
There is a reason why I abandoned libprussdrv after validating the concept - the fact that the shared memory held by libprussdrv is allocated on the kernel side through dma_alloc_coherent. This means that every memory access is direct and *through* the cache, i.e. the cache is not utilized, hence you see slow read speeds.

Unfortunately there's no way to get higher speeds than to get in to the kernel and allocate your own memory - that's what BeagleLogic does now. Have a look at the kernel module code to see how it works.

credentiality, 2016-01-10:
Thanks, that's very helpful!

abhishek-kakkar, 2016-01-10:
Closing this issue as this does not affect BeagleLogic.

For discussions like this in the future, you might want to use the newly created Google Group at https://groups.google.com/forum/#!forum/beaglelogic .

closed 2016-01-10 by abhishek-kakkar