abhishek-kakkar/BeagleLogic #12: beaglelogic-20150817.img doesn't serve DHCP over USB

credentiality, 2016-02-17:
I kept trying different SD cards with the image because the board never came up with the expected 192.168.7.1 address via USB, but after many hours I noticed eth1 was showing up and was showing traffic.  tcpdump revealed the board was trying to get a DHCP address instead of serving 192.168.7.1 to the host.  'ifconfig eth1 192.168.7.1' on the host side allowed me to ssh in.

abhishek-kakkar, 2016-02-17:
This will be fixed in a subsequent image release (ETA: the coming week)

credentiality, 2016-02-17:
Cool.  BTW, I also get:

debian@beaglebone:/opt/BeagleLogic/beaglelogic-firmware$ sudo make
  CC	beaglelogic-pru0.obj
"pru_syscalls.h", line 10: fatal error: cannot open source file "stdlib.h"
1 catastrophic error detected in the compilation of "beaglelogic-pru0.c".
Compilation terminated.

>> Compilation failure
Makefile:41: recipe for target 'beaglelogic-pru0.obj' failed
make: *** [beaglelogic-pru0.obj] Error 1

abhishek-kakkar, 2016-02-17:
Check environment variable PRU_C_DIR as mentioned here - https://github.com/abhishek-kakkar/BeagleLogic/wiki/Build-BeagleLogic

closed 2016-03-08 by abhishek-kakkar