abhishek-kakkar/BeagleLogic #13: Use PRU0 for input

NiHaoMike, 2016-05-15:
How difficult would it be to modify BeagleLogic to use PRU0 for input? That way, the HDMI can still be used at the same time, possibly allowing for a GUI running on the unit itself.

abhishek-kakkar, 2016-05-17:
The sample rate would be affected - 50 Msps would be the maximum theoretically possible sample rate with a single PRU approach (taking assembly instructions - sample, store, increment and loop jump).

I understand the desire to use HDMI - it was something I have wanted to do myself but then the way the BeagleBone hardware is configured - you can't really reassign PRU1 pins somewhere else and get to use the LCD interface together with PRU1. If you really wanted a LCD you could attach it over the SPI bus and use it with the fbtft kernel driver.

The BeagleBone may be good at capturing logic data but processing on it is much slower than on, say, a laptop - a web interface can take advantage of the processing power of the PC.

This discussion can be further continued on the BeagleLogic discussion group.

NiHaoMike, 2016-05-19:
Could PRU0 handle the input and PRU1 the buffering? In other words, have PRU1 do what PRU0 does and vice versa?

abhishek-kakkar, 2020-07-17:
Role reversal is useful for implementing BeagleLogic on new platforms such as the PocketBeagle. This will be implemented soon. Will close the issue as I consolidate everything into one issue.

closed 2020-07-17 by abhishek-kakkar