abhishek-kakkar/BeagleLogic #15: functionality request

skypuppy, 2016-10-24:
*** Please set up a place where we can discuss, ask questions, and generally bug you.  :)  For example, my questions are: 1) is the cape a must in order to use this great software?  2) I would love a method of using a host Linux (or if absolutely necessary) a Winbloze laptop, in order to use PulseView for continuous sampling of the UART (serial) and the I2C bus, but I need to have it run many minutes, exceeding the BBB's memory. In other words, let the desktop do the heavy lifting and just let the BBB capture the data and send it (unprocessed?) via ethernet (pref) or USB.  I'm told libsigrok can accommodate that but I can't do that with a Bus Pirate.  A forum or chat room would work, pref.a forum so we can look back over history.  Thanks!

abhishek-kakkar, 2016-10-24:
> Please set up a place where we can discuss, ask questions, and generally bug you. :)

There already is - [Google Groups for BeagleLogic](https://groups.google.com/forum/#!forum/beaglelogic)

> 1) is the cape a must in order to use this great software? 

Not if all you have is 3.3V logic signals, in that case it is OK to connect inputs from P8_39 to P8_46 on the BeagleBone hardware directly

> 2) I would love a method of using a host Linux (or if absolutely necessary) a Winbloze laptop, in order to use PulseView for continuous sampling of the UART (serial) and the I2C bus, but I need to have it run many minutes, exceeding the BBB's memory.

I am out of development bandwidth at the moment so cannot develop a permanent solution at this point (though it will the first thing to look at when I do get the time). In the meantime please use a combination of dd'ing from /dev/beaglelogic and netcat over the network ( whether it is the USB virtual Ethernet or a LAN connection ), hope you get the idea.

Use the forum to discuss any further questions.

skypuppy, 2016-10-24:
Google can't find my password and/or does not email the link to reset that password.  I've tried several times (for this and other projects.)  And as you well know, Google people do not respond to any of it's users.  So I have no method of even logging in to ANY Google forum.  Would you like me to see if I can host a forum on one of my machines?
Will netcat allow a continuous dump or just one sample segment at a time?  I've never used netcat or ncat and their wiki entries are unclear.

abhishek-kakkar, 2016-10-26:
I'm sorry about your Google account thing but it's not necessary to have a Gmail ID to start a discussion or even view the archives (the BeagleLogic forum is configured to be publicly accessible even if you are not signed in). You can always send an email to beaglelogic@googlegroups.com to create a new thread and operate mailing-list style.

To switch between one-shot and continous sampling modes, the 'triggerflags' attribute needs to be set. Refer to the Wiki page on [sysfs attributes](https://github.com/abhishek-kakkar/BeagleLogic/wiki/sysfs-attributes-Reference). In one-shot mode there will be an EOF generated when the buffer is completely read out, in case of continuous mode, sampling continues until the file is closed.

abhishek-kakkar, 2017-10-07:
BeagleLogic now has a TCP backend (see tcp-server-node/ in the code) and can be used via PulseView in Linux currently. See: https://hackaday.io/project/25745-beaglelogic-standalone/log/67652-major-change-in-how-beaglelogic-can-be-used

abhishek-kakkar, 2017-10-07:
Closing this as it's been a while now. Feel free to reopen if needed.

closed 2017-10-07 by abhishek-kakkar