abhishek-kakkar/BeagleLogic #16: input voltage to the BBB

skypuppy, 2016-12-21:
AIUI, none of the input pins on the BBB are allowed more than 1.8V.  How do sample most of the world?  Even most uprocessors output 3.3V, and then there is as much as 12v on full-blown RS-232.  How do we not let the smoke out of the BBB, especially when we're not sure of the voltage we're trying to sample???

pro585code, 2016-12-21:
Test the voltage lvl first... Then use proper voltage divider to step down that voltage while trying to keep you signal intact?

pro585code, 2016-12-21:
Your*

ItsMeMegaMind, 2017-06-17:
Is there a way we can make an external ADC work with beaglelogic? That would solve the problem.

abhishek-kakkar, 2017-10-07:
Google "PRUDAQ" for an external ADC interfacing with BeagleLogic.

I have designed a cape for the BeagleBone that allows safe interfacing of up to 5V logic level signals, and will work on enabling signal levels > 5V soon. Please watch the wiki page for announcements.

Closing this issue.

closed 2017-10-07 by abhishek-kakkar