abhishek-kakkar/BeagleLogic #17: Compatible with Beaglebone Black Wireless?

dlee1579, 2017-03-10:
I've been having MAJOR problems with installing PRUDAQ and BeagleLogic onto my BBB. I just now realized that all of this was written for BBB and not specifically BBB Wireless. Is this where my problems are stemming from? For instance, I can't get the PRUDAQ BeagleLogic image to boot onto my BBBW.

skypuppy, 2017-03-10:
Do you have a gen 1 BBB that you can try the SD card with?  Maybe the 
image or the card is not wonderful.

dlee1579, 2017-03-11:
Currently I do not. I was also thinking the image may have been problematic.

For what it's worth, I've burned the image onto a 32G sd card on my
Macbook, and I've tried multiple methods of burning onto the card: Etcher
and sudo dd in terminal. When I burn Luc's image onto the sd card, I cannot
even see what is on the card. When my friend loaded the sd card on his
linux-based laptop, he could see all the contents on the card.

skypuppy, 2017-03-11:
Sounds like a couple of potential problems.  First, SD card might be 
bad.  Second, the downloaded image might be suspect.

Where are you?  If it is practical you could send me your card (or 
image) and I could try it in a gen 1 BBB  as I have four of those.  So 
far, I see no definitive resolution to your problem.

David

hlipka, 2017-03-24:
I have now both at home (a BBB Rev B, and a BBB wireless). I have a known-good image (the 'no-setup-image', booting at the BBB), and it doesn't work in the BBBW. I attached a FTDI adapter to the serial header, and this is what I get during boot:

1227 bytes read in 48 ms (24.4 KiB/s)
debug: [/boot/vmlinuz-3.8.13-bone79] ...
5584528 bytes read in 423 ms (12.6 MiB/s)
debug: [/boot/initrd.img-3.8.13-bone79] ...
3670304 bytes read in 299 ms (11.7 MiB/s)
debug: [/boot/dtbs/3.8.13-bone79/am335x-boneblack-wireless.dtb] ...
** File not found /boot/dtbs/3.8.13-bone79/am335x-boneblack-wireless.dtb **
debug: [console=tty0 console=ttyO0,115200n8 capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN root=/dev/mmcblk0p1 rootfstype=ext4 rootwait coherent_pool=1M quiet cape_universal=enable] ...
debug: [bootz 0x82000000 0x88080000:380120 0x88000000] ...
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
Checking if client_ip is set ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
1227 bytes read in 48 ms (24.4 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-3.8.13-bone79 ...
5584528 bytes read in 423 ms (12.6 MiB/s)

unable to find [dtb=am335x-boneblack-wireless.dtb] did you name it correctly? ...

FAILSAFE: U-Boot UMS (USB Mass Storage) enabled, media now available over the usb slave port ...

----------------------------------
Whats interesting: I have the same image on a 16Gb SD card, and it results in a different boot trace:

1227 bytes read in 53 ms (22.5 KiB/s)

debug: [/boot/vmlinuz-3.8.13-bone79] ...

5584528 bytes read in 360 ms (14.8 MiB/s)

debug: [/boot/initrd.img-3.8.13-bone79] ...

3670304 bytes read in 259 ms (13.5 MiB/s)

debug: [/boot/dtbs/3.8.13-bone79/am335x-boneblack.dtb] ...

26118 bytes read in 79 ms (322.3 KiB/s)

debug: [console=tty0 console=ttyO0,115200n8 capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN root=/dev/mmcblk0p1 rootfstype=ext4 rootwait coherent_pool=1M quiet cape_universal=enable] ...

debug: [bootz 0x82000000 0x88080000:380120 0x88000000] ...

Kernel image @ 0x82000000 [ 0x000000 - 0x553690 ]

## Flattened Device Tree blob at 88000000

   Booting using the fdt blob at 0x88000000

   Using Device Tree in place at 88000000, end 88009605

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
[    0.432919] omap2_mbox_probe: platform not supported
[    0.591615] tps65217-bl tps65217-bl: no platform data provided
[    0.654432] bone-capemgr bone_capemgr.9: slot #0: No cape found
[    0.691542] bone-capemgr bone_capemgr.9: slot #1: No cape found
[    0.728649] bone-capemgr bone_capemgr.9: slot #2: No cape found
[    0.765758] bone-capemgr bone_capemgr.9: slot #3: No cape found
[    0.784399] omap_hsmmc mmc.5: of_parse_phandle_with_args of 'reset' failed
[    0.846093] pinctrl-single 44e10800.pinmux: pin 44e10854 already requested by 44e10800.pinmux; cannot claim for gpio-leds.8
[... 20 lines cut ...]
[    1.048418] 1e60: 00000001 c0108a6d df0494b8 c004bab7 00000000 df0d5444 df0d9c48 00000000
[    1.056965] 1e80: df0d5410 df0d5410 00000000 c0945dec df0d5410 df0d5444 c08b4bc0 c08b4bc0
[    1.065513] 1ea0: 00000102 c0836328 c082af49 c02bf789 df0d5410 c08b4bc0 df0d5444 c08b4bc0
[    1.074058] 1ec0: c02bf8d9 c02bf923 c08b4bc0 00000000 00000001 c02be677 df049478 df0c6180
[    1.082605] 1ee0: c08b4bc0 de52a7c0 c08aa1b8 00000000 c08d6980 c02bf1a7 c07542d4 c08b4bc0
[    1.091151] 1f00: df070010 c08b4bc0 df070010 c0836324 c08d6980 c02bfd0d c08d6980 df070010
[    1.099699] 1f20: c0836324 c08d6980 c08d6980 c0008757 00000000 c0046175 c07d40b8 00000000
[    1.108246] 1f40: 00000007 c0d5ebb3 00000000 c0849730 00000007 c0836324 c08d6980 c08d6980
[    1.116793] 1f60: 00000102 c0836328 00000000 c080d755 00000007 00000007 c080d1f9 c004f439
[    1.125338] 1f80: 00000000 c04cf22d 00000000 00000000 00000000 00000000 00000000 c04cf233
[    1.133883] 1fa0: 00000000 00000000 c04cf22d c000c879 00000000 00000000 00000000 00000000
[    1.142428] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.150972] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[    1.159538] [<c031dc28>] (cpsw_probe+0x37c/0x9e0) from [<c02bf789>] (driver_probe_device+0x55/0x180)
[    1.169087] [<c02bf789>] (driver_probe_device+0x55/0x180) from [<c02bf923>] (__driver_attach+0x4b/0x4c)
[    1.178907] [<c02bf923>] (__driver_attach+0x4b/0x4c) from [<c02be677>] (bus_for_each_dev+0x3b/0x5c)
[    1.188363] [<c02be677>] (bus_for_each_dev+0x3b/0x5c) from [<c02bf1a7>] (bus_add_driver+0x77/0x184)
[    1.197819] [<c02bf1a7>] (bus_add_driver+0x77/0x184) from [<c02bfd0d>] (driver_register+0x3d/0xc8)
[    1.207186] [<c02bfd0d>] (driver_register+0x3d/0xc8) from [<c0008757>] (do_one_initcall+0xcb/0x10c)
[    1.216651] [<c0008757>] (do_one_initcall+0xcb/0x10c) from [<c080d755>] (kernel_init_freeable+0x10d/0x180)
[    1.226747] [<c080d755>] (kernel_init_freeable+0x10d/0x180) from [<c04cf233>] (kernel_init+0x7/0x10c)
[    1.236392] [<c04cf233>] (kernel_init+0x7/0x10c) from [<c000c879>] (ret_from_fork+0x11/0x38)
[    1.245211] Code: f000 8193 f8c4 05c0 (6803) f44f 
[    1.250221] ---[ end trace 5be192abe05c1ef6 ]---
[    1.255240] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

hlipka, 2017-03-24:
So it seems the 3.8 kernel on that image is not compatible with the new processor on the BBBW. What would it take to get BeagleLogic running on a 4.4 kernel? (Right now I'm trying to compile the current sources on my BBB, but no luck so far).

abhishek-kakkar, 2017-03-25:
The BBW was released about the same time as this image was built (July 2016), so it may be that a newer build of the image may be needed.

This may need a new image build.

hlipka, 2017-03-25:
The current image uses the kernel 3.8, and I don't the base images with that kernel have been updated for the BBBW.

abhishek-kakkar, 2017-03-25:
The base images run 4.4 yes, the image build switches the kernel to 3.8.13. If you can try installing BeagleLogic / PRUDAQ on the latest available image it should work (not sure about wireless functionality working though).

hlipka, 2017-03-25:
When I understand https://github.com/abhishek-kakkar/BeagleLogic/wiki/Build-BeagleLogic correctly, the BeagleLogic modules should be already in the image, right? That would mean I need only the firmware.
Unfortunately https://github.com/abhishek-kakkar/BeagleLogic/wiki/BeagleLogic-Primer-:-Getting-Started says that BL is _not_ compatible with 4.4 kernels...
Would the base image one from the official page: http://beagleboard.org/latest-images (I would use Jessie IoT (non-GUI) version 8.7 since HDMI is disabled anyway)? V8.6 is the oldest with BBB Wireless support.
I would follow the wiki, starting at "build beaglelogic".

abhishek-kakkar, 2017-03-25:
On a new image running kernel 4.4, running sudo apt-get update & sudo install linux-image-3.8.13-* ( do an apt-cache search to find the latest version) should install the kernel and modules for you and also set the boot kernel to 3.8.13.

Then you just need to copy the beaglelogic firmware built to /lib/firmware .

hlipka, 2017-03-25:
Nope, kernel 3.8.13-bone84 (which is the newest one thats available when using the above image) doesn't play nice with the BBBW:
(serial console log)
Uncompressing Linux... done, booting the kernel.␍␊
[    0.449922] omap2_mbox_probe: platform not supported␍␊
[    0.608684] tps65217-bl tps65217-bl: failed to find backlight node␍␊
[    0.608698] tps65217-bl tps65217-bl: DT parse failed.␍␊
[    0.677567] bone-capemgr bone_capemgr.9: slot #0: No cape found␍␊
[    0.714675] bone-capemgr bone_capemgr.9: slot #1: No cape found␍␊
[    0.751784] bone-capemgr bone_capemgr.9: slot #2: No cape found␍␊
[    0.788893] bone-capemgr bone_capemgr.9: slot #3: No cape found␍␊
[    0.804415] bone-capemgr bone_capemgr.9: slot #6: BB-BONELT-HDMIN conflict P8.45 (#5:BB-BONELT-HDMI)␍␊
[    0.814039] bone-capemgr bone_capemgr.9: slot #6: Failed verification␍␊
[    0.820785] bone-capemgr bone_capemgr.9: loader: failed to load slot-6 BB-BONELT-HDMIN:00A0 (prio 2)␍␊
[    0.837320] omap_hsmmc mmc.5: of_parse_phandle_with_args of 'reset' failed␍␊
[    0.900500] pinctrl-single 44e10800.pinmux: pin 44e10854 already requested by 44e10800.pinmux; cannot claim for gpio-leds.8␍␊
[    0.912167] pinctrl-single 44e10800.pinmux: pin-21 (gpio-leds.8) status -22␍␊
[    0.919452] pinctrl-single 44e10800.pinmux: could not request pin 21 on device pinctrl-single␍␊
Loading, please wait...␍␊
rootfs: clean, 114435/217728 files, 775567/869376 blocks␍␊
[  120.601424] INFO: task mmcqd/0:72 blocked for more than 60 seconds.␍␊
[  120.608001] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.␍␊
[  120.616218] Kernel panic - not syncing: hung_task: blocked tasks␍␊
[  120.622516] [<c0011611>] (unwind_backtrace+0x1/0x90) from [<c04d325f>] (panic+0x73/0x174)␍␊
[  120.631061] [<c04d325f>] (panic+0x73/0x174) from [<c0078401>] (watchdog+0x1d9/0x1f0)␍␊
[  120.639161] [<c0078401>] (watchdog+0x1d9/0x1f0) from [<c0047577>] (kthread+0x83/0x84)␍␊
[  120.647346] [<c0047577>] (kthread+0x83/0x84) from [<c000c879>] (ret_from_fork+0x11/0x38)␍␊
[  120.655798] drm_kms_helper: panic occurred, switching back to text console␍␊

hlipka, 2017-03-25:
OTOH: this also happens when I boot the same image with the 3.8.13-bone84 kernel on my regular BBB, so this might be a different issue. I need to get a 4 or 8GB SD card, they seem to be more reliable, my 16GB card acts funny.
Can we move that discussion to email or the forum, to keep this issue for the real problem (or the solution, hopefully)?

abhishek-kakkar, 2017-06-05:
I am currently working to port BeagleLogic to work with kernel version 4.9. Once this is done, hopefully BeagleLogic should also be able to run on Seeed BeagleBone Green Wireless.

abhishek-kakkar, 2017-10-07:
With BeagleLogic being ported to kernel version 4.9, this has been fixed. Closing.

closed 2017-10-07 by abhishek-kakkar