abhishek-kakkar/BeagleLogic #19: Port to Kernel 4.4 and above

abhishek-kakkar, 2017-06-05:
BeagleLogic, as written in 2014, used the 3.8.13 kernel release. This kernel version is quite dated as of 2017 and hence it's time that BeagleLogic evolve and keep up with the latest kernel versions.

I am currently done modifying the PRU firmware, and currently porting the BeagleLogic kernel driver to work with the PRU remoteproc driver in kernel version 4.9 (TI). This issue will track the progress of the work.

sarnold, 2017-07-04:
Since you didn't mention this, I kinda assumed it was working.  I made a card like you said, but forgot to update the kernel, did that and re-installed beaglelogic, but still no firmware or /dev/beaglogic.  Note the PRUs are "available" but never loaded:

```
$ dmesg | tail
[  257.379473] c_can_platform 481d0000.can: c_can_platform device registered (regs=fa1d0000, irq=170)
[  260.201119] omap_rng 48310000.rng: OMAP Random Number Generator ver. 20
[  261.292776] ti-pruss 4a300000.pruss: creating PRU cores and other child platform devices
[  261.414638] remoteproc remoteproc1: 4a334000.pru0 is available
[  261.414811] pru-rproc 4a334000.pru0: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a334000 probed successfully
[  261.444464] remoteproc remoteproc2: 4a338000.pru1 is available
[  261.444678] pru-rproc 4a338000.pru1: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a338000 probed successfully
[  261.456559] omap-sham 53100000.sham: hw accel on OMAP rev 4.3
[  261.493971] omap-aes 53500000.aes: OMAP AES hw accel rev: 3.2
[  261.500329] omap-aes 53500000.aes: will run requests pump with realtime priority
```

The module only loads if I put it in /etc/modules and there is still no device:

```
$ ls /dev/b*
/dev/binder  /dev/btrfs-control
/dev/block:
179:0  179:1  179:24  179:32  179:8  179:9
/dev/bus:
usb
```

abhishek-kakkar, 2017-07-05:
Can you post the output of:
journalctl | grep beaglelogic

and uEnv.txt?

sarnold, 2017-07-05:
Too late, I reflashed the card to try "fresh" without u-boot overlays, but I haven't run your install script yet, just added the firmware/dtbo files and trying capemgr enable/disable.  The previous output was from your install.sh (ie, unedited uEnv.txt).

sarnold, 2017-07-05:
without u-boot overlays enabled, it all "appears" to work except I get no green enable led on prudaq.  This is after making/installing both beaglelogic and prudaq firmware, rebooting with only emmc-overlay.dtb and cape-universalh enabled, then running the startup script manually.  After running the script I see this in dmesg:

(note that beaglelogic-pru1-fw is a symlink to the beaglelogic-pru1-prudaq-ch01 firmware)

```
[  254.733081] ti-pruss 4a300000.pruss: creating PRU cores and other child platform devices
[  254.877206] remoteproc remoteproc1: 4a334000.pru0 is available
[  254.877375] pru-rproc 4a334000.pru0: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a334000 probed successfully
[  254.917689] remoteproc remoteproc2: 4a338000.pru1 is available
[  254.946645] pru-rproc 4a338000.pru1: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a338000 probed successfully
[  255.261407] omap-sham 53100000.sham: hw accel on OMAP rev 4.3
[  255.577187] omap-aes 53500000.aes: OMAP AES hw accel rev: 3.2
[  255.595426] omap-aes 53500000.aes: will run requests pump with realtime priority
[  271.055200] bone_capemgr bone_capemgr: part_number 'beaglelogic', version 'N/A'
[  271.055236] bone_capemgr bone_capemgr: slot #5: override
[  271.065300] bone_capemgr bone_capemgr: Using override eeprom data at slot 5
[  271.065320] bone_capemgr bone_capemgr: slot #5: 'Override Board Name,00A0,Override Manuf,beaglelogic'
[  271.093911] remoteproc remoteproc1: powering up 4a334000.pru0
[  271.099758] remoteproc remoteproc1: Booting fw image beaglelogic-pru0-fw, size 62976
[  271.104439] ti-pruss 4a300000.pruss: configured system_events = 0x0000000001d30000 intr_channels = 0x00000037 host_intr = 0x00000037
[  271.104467] remoteproc remoteproc1: remote processor 4a334000.pru0 is now up
[  271.104544] remoteproc remoteproc2: powering up 4a338000.pru1
[  271.107896] remoteproc remoteproc2: Booting fw image beaglelogic-pru1-fw, size 27580
[  271.107972] remoteproc remoteproc2: remote processor 4a338000.pru1 is now up
[  271.108023] BeagleLogic loaded and initializing
[  271.112849] misc beaglelogic: Valid PRU capture context structure found at offset 0000
[  271.112872] misc beaglelogic: BeagleLogic PRU Firmware version: 0.3
[  271.112881] misc beaglelogic: Device supports max 128 vector transfers
[  271.112905] misc beaglelogic: Default sample rate=50000000 Hz, sampleunit=1, triggerflags=0. Buffer in units of 4194304 bytes each
[  271.113194] bone_capemgr bone_capemgr: slot #5: dtbo 'beaglelogic-00A0.dtbo' loaded; overlay id #1
[  272.606050] misc beaglelogic: Successfully allocated 67108864 bytes of memory.
```

sarnold, 2017-07-05:
Sorry, it's the script from the prudaq repo that finally enables, but the script doesn't work without commenting out the 4 echos at the bottom.  Why? becuase it tries to add an overlay for a pwm device in an ocp path that does not exist.  Running the pin tweaks above those 4 commands does light the enable led, but without setting the pwm stuff there is no capture.  So which pwm device under /sys/devices/platform/ocp/ do we need to setup?  Do we need another overlay to replace the missing bone_pwm_P9_31 ?  That should probably be the answer I think.  Unless you have another way for me to try without that?

sarnold, 2017-07-05:
Take a look at pull request #21, maybe it's helpful...

sarnold, 2017-07-05:
Actually, the pull request is superfluous, since it looks like this is the way for now:

https://github.com/beagleboard/bb.org-overlays/blob/master/examples/cape-unversal-pwm.txt

Maybe you can see the right way to connect one of these with beaglelogic/prudaq?

abhishek-kakkar, 2017-10-07:
BeagleLogic has been successfully ported to kernel version 4.9. Hence closing this.

closed 2017-10-07 by abhishek-kakkar