abhishek-kakkar/BeagleLogic #26: Upgrading prebuilt image

Vagabond, 2017-11-02:
I've tried to update the git repo in the prebuilt image (both with simply updating git and also running install.sh) and it seems to break things badly. Is this still expected to work?

abhishek-kakkar, 2017-11-03:
Did you run install.sh as-is or "install.sh --upgrade" (see http://docs.beaglelogic.net/en/latest/install.html#upgrading-an-existing-installation)?

Running install.sh as-is modifies /boot/uEnv.txt and if you run it more than once, your /boot/uEnv.txt needs fixing.

Can you also describe as to what is broken?

Vagabond, 2017-11-08:
Sorry, I didn't have my beaglebone over the weekend.

First thing is that a `git pull` doesn't actually work out of the box:

```
debian@beaglebone:/opt/BeagleLogic$ sudo git pull
remote: Counting objects: 425, done.
remote: Total 425 (delta 129), reused 129 (delta 129), pack-reused 296
Receiving objects: 100% (425/425), 82.72 KiB | 0 bytes/s, done.
Resolving deltas: 100% (267/267), completed with 16 local objects.
From https://github.com/abhishek-kakkar/BeagleLogic
   2cdc9c6..3cea157  master     -> origin/master
 * [new branch]      4.4-dev    -> origin/4.4-dev
 * [new branch]      standalone -> origin/standalone
Auto-merging kernel/beaglelogic.h
Auto-merging kernel/beaglelogic.c
Auto-merging firmware/beaglelogic-pru1-core.asm
Auto-merging firmware/beaglelogic-pru0-core.asm
Auto-merging firmware/beaglelogic-pru-defs.inc
Removing beaglelogic-kernel-driver/beaglelogic_glue.h
CONFLICT (modify/delete): beaglelogic-kernel-driver/Makefile deleted in 3cea157d312a685547a2ca9aa66cefacd1fd2cee and modified in HEAD. Version HEAD of beaglelogic-kernel-driver/Makefile left in tree.
Removing beaglelogic-kernel-driver/BB-BEAGLELOGIC-00A0.dts
Removing beaglelogic-firmware/pru_syscalls.h
Removing beaglelogic-firmware/pru_syscalls.asm
Removing beaglelogic-firmware/pru_defs.h
Removing beaglelogic-firmware/beaglelogic-pru0.c
Removing beaglelogic-firmware/beaglelogic-pru0-syscalls.asm
Removing beaglelogic-firmware/am335x_pru.cmd
Removing beaglelogic-firmware/Makefile
Automatic merge failed; fix conflicts and then commit the result.
```

I did a git reset --hard origin/master to get the latest git code.

Then, when I run the install --upgrade, I get several 'catastrophic' errors:

```
debian@beaglebone:/opt/BeagleLogic$ sudo sh install.sh --upgrade
beaglelogic: Building and installing PRU firmware

************************************************************
Building project: firmware

Building file: beaglelogic-pru0.c
Invoking: PRU Compiler
/usr/share/ti/cgt-pru/bin/clpru --include_path=/usr/share/ti/cgt-pru/include --include_path=/usr/lib/ti/pru-software-support-package/include --include_path=/usr/lib/ti/pru-software-support-package/include/am335x -v3 -O2 --display_error_number --endian=little --hardware_mac=on --obj_directory=release --pp_directory=release -ppd -ppa -fe release/beaglelogic-pru0.object beaglelogic-pru0.c
"beaglelogic-pru0.c", line 18: fatal error #1965: cannot open source file "pru_cfg.h"
1 catastrophic error detected in the compilation of "beaglelogic-pru0.c".
Compilation terminated.

>> Compilation failure
Makefile:101: recipe for target 'release/beaglelogic-pru0.object' failed
make: *** [release/beaglelogic-pru0.object] Error 1

Building file: beaglelogic-pru0.c
Invoking: PRU Compiler
/usr/share/ti/cgt-pru/bin/clpru --include_path=/usr/share/ti/cgt-pru/include --include_path=/usr/lib/ti/pru-software-support-package/include --include_path=/usr/lib/ti/pru-software-support-package/include/am335x -v3 -O2 --display_error_number --endian=little --hardware_mac=on --obj_directory=release --pp_directory=release -ppd -ppa -fe release/beaglelogic-pru0.object beaglelogic-pru0.c
"beaglelogic-pru0.c", line 18: fatal error #1965: cannot open source file "pru_cfg.h"
1 catastrophic error detected in the compilation of "beaglelogic-pru0.c".
Compilation terminated.

>> Compilation failure
Makefile:101: recipe for target 'release/beaglelogic-pru0.object' failed
make: *** [release/beaglelogic-pru0.object] Error 1

************************************************************
Building project: prudaq

Building file: beaglelogic-pru1.c
Invoking: PRU Compiler
[... 146 lines cut ...]
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libaudio2 libmng1 libqt4-network libqt4-xml libqtcore4 libqtdbus4 libqtgui4
  qtcore4-l10n
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  golang-doc golang-go golang-go-linux-arm golang-go.tools golang-src
The following NEW packages will be installed:
  golang golang-doc golang-go golang-go-linux-arm golang-go.tools golang-src
0 upgraded, 6 newly installed, 0 to remove and 11 not upgraded.
Need to get 26.9 MB of archives.
After this operation, 163 MB of additional disk space will be used.
Get:1 http://httpredir.debian.org/debian/ jessie/main golang-src armhf 2:1.3.3-1 [5,143 kB]
Get:2 http://httpredir.debian.org/debian/ jessie/main golang-go-linux-arm armhf 2:1.3.3-1 [3,361 kB]
Get:3 http://httpredir.debian.org/debian/ jessie/main golang-go armhf 2:1.3.3-1 [8,193 kB]
Get:4 http://httpredir.debian.org/debian/ jessie/main golang-doc all 2:1.3.3-1 [1,950 kB]
Get:5 http://httpredir.debian.org/debian/ jessie/main golang all 2:1.3.3-1 [25.0 kB]
Get:6 http://httpredir.debian.org/debian/ jessie/main golang-go.tools armhf 0.0~hg20140703-4 [8,214 kB]
Fetched 26.9 MB in 14s (1,797 kB/s)
beaglelogic: Successfully Installed. Please reboot
```

I also see several errors that get overwritten about no space left on device, so it is clearly running out of room on the SD card.

Vagabond, 2017-11-08:
And, after rebooting as instructed, `systemctl -l` shows both the beaglebone-tcp.service and the lightdm.service in a failed state.

Additionally, no services are listening on port 4000 anymore, either.

Vagabond, 2017-11-08:
I did a partition/filesystem resize with fdisk and resize2fs, so there's space now, and things work a bit better. However, I still have the 2 previously mentioned failed services, but now there's a third one called beaglelogic.service.

Additionally, it looks like the `beaglelogic` user/group does not exist, and it is only added in non-upgrade mode. The npm install tries to use the 'debian' (or maybe the beaglelogic) user to install with, but it is failing because of the root ownership of /opt/BeagleLogic.

I was able to, I think, fix things by doing `npm install` and the `go build` as root in the appropriate directories. The services are running again, at least.

abhishek-kakkar, 2017-11-09:
Ah I see why you are getting the error now. You're using the prebuilt image from the wiki that's based on the 3.8.13 kernel.

You have to switch to a new prebuilt image available at : http://docs.beaglelogic.net/en/latest/beaglelogic_system_image.html . This is based on the 4.9 kernel version and is a **major** upgrade from the 3.8.13 system image.

Vagabond, 2017-11-09:
Ah, I see. The older image is the one that I found a download link for. Thanks for the help.

abhishek-kakkar, 2017-11-09:
Let me know when you are able to run the new image successfully and able to sample data so that I can close the issue.

Vagabond, 2017-11-09:
It seems to work, I think we can close this.

closed 2017-11-09 by Vagabond