abhishek-kakkar/BeagleLogic #28: PocketBeagle support

AnthonyDiGirolamo, 2017-12-23:
Is it possible to run BeagleLogic on the PocketBeagle?  https://github.com/beagleboard/pocketbeagle/wiki/System-Reference-Manual

NeoFromMatrix, 2019-03-22:
Seems like this is pretty much the same SoC as used for https://hackaday.io/project/25745-beaglelogic-standalone , the PocketBeagle is obviously missing the 74series chip and also the eeprom. 

Did you try to compile it on your own with a kernel version of 4.9 or lower (to hopefully avoid #39) ?

abhishek-kakkar, 2020-07-04:
@AnthonyDiGirolamo -- Yes, it is, although PocketBeagle will be limited to only 8 channels. It requires "special treatment", I am working out a pathway through which it is possible, because PocketBeagle has the PRU0 Inputs exposed, BeagleLogic normally uses PRU1, so it will be required for the PRUs to "swap roles". This is also required to get BeagleLogic working on the BeagleBone AI.

You should hear more on this in the coming week or two. Thanks for bearing with me during this time.

Rkk-10, 2022-10-09:
Is it ready?