abhishek-kakkar/BeagleLogic #30: Add PulseView to image

buzmeg, 2018-02-12:
PulseView is pretty difficult to build since it needs so much cross-compilation expertise and the dependency chain through sigrok is quite long.

Please add an image of PulseView to the BeagleLogic image on the next build.

Thanks.

abhishek-kakkar, 2018-02-12:
In the background I have added a TCP/IP backend to BeagleLogic, so that one can use BeagleLogic with PulseView runnning on the desktop and connecting to the hardware (BeagleBone or the BeagleLogic Standalone board) via TCP/IP. So one does not, in theory, need to run PulseView on the same hardware running BeagleLogic. I have to add instructions on how to use this.

buzmeg, 2020-02-21:
Project dead.  Closing.

closed 2020-02-21 by buzmeg