abhishek-kakkar/BeagleLogic #31: PRU probe fails on SanCloud Beaglebone Enhanced

timothy-chapman, 2018-05-23:
Kumar,

I realize the BeagleLogic has not been tested on the SanCloud BeagleBone Enhanced. However, we are trying anyway to get your latest system image (beaglelogic-stretch-2017-07-13-4gb.img.xz) running. 

We are at the point in your suggested Troubleshooting (https://beaglelogic.readthedocs.io/en/latest/beaglelogic_system_image.html) where we are advised to open an issue.

> debian@beaglebone:~$ lsmod | grep logic
> beaglelogic            18177  0
> pruss_intc              9092  2 pru_rproc,beaglelogic
> pruss                  10889  2 pru_rproc,beaglelogic

OK so far.

But /dev/beaglelogic does not exist and the output from 'journalctl | grep beaglelogic' does not look as it should:

debian@beaglebone:~$ journalctl | grep beaglelogic
Jul 13 08:27:25 beaglebone bash[865]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Jul 13 08:27:51 beaglebone beaglelogic[838]: Express server listening on port 4000
Jul 13 08:28:21 beaglebone kernel: beaglelogic pru-beaglelogic: Unable to get pruss handle.
Jul 13 08:28:21 beaglebone kernel: beaglelogic: probe of pru-beaglelogic failed with error -2
Jul 13 08:29:29 beaglebone bash[865]: beaglelogic-startup: timeout. BeagleLogic couldn't load.
Jul 13 08:29:29 beaglebone systemd[1]: beaglelogic-startup.service: Main process exited, code=exited, status=1/FAILURE
Jul 13 08:29:29 beaglebone systemd[1]: beaglelogic-startup.service: Unit entered failed state.
Jul 13 08:29:29 beaglebone systemd[1]: beaglelogic-startup.service: Failed with result 'exit-code'.

Note: The actual date was 2018-05-23--we just quit setting time on the bone. 

Please advise.

Thanks in advance

abhishek-kakkar, 2018-05-24:
I wonder if this is a Device Tree issue. Maybe the nodes that refer to PRUSS are not loaded.

I'd need more kernel logs to look into this.

timothy-chapman, 2018-05-24:
Thanks so much.
The attached file is the output from 'dmesg' on the SanCloud machine ASAP after booting the latest BeagleLogic image.
[dmesg.txt](https://github.com/abhishek-kakkar/BeagleLogic/files/2036529/dmesg.txt)

timothy-chapman, 2018-05-26:
The above capture should have been called "dmesg.sancloud.txt".
This one was acquired for comparison--same image file--booted on a "BBB Rev C"
[dmesg.bbb-revc.txt](https://github.com/abhishek-kakkar/BeagleLogic/files/2041240/dmesg.bbb-revc.txt)

sancloud, 2018-05-29:
I have been having a look at this and it seems to be something to do with overlays, either in uboot or the kernel.  If I program the serial number on one of our boards to be the same as a BBB the system comes up OK.
I will have more of a play with the kernel device tree before I delve into uboot.

Will update when I have more info.

sancloud, 2018-05-29:
Ok I have found the issue.  It is to do with the uboot device tree.
For the BBB the pru is enabled and for the BBE it is disabled (for some reason).

If I modify the uboot.dtb and boot:

[   59.866903] usbcore: registered new interface driver rtl8xxxu
[   59.899754] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   75.919714] omap_rng 48310000.rng: OMAP Random Number Generator ver. 20
[   77.232037] ti-pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   77.251111] remoteproc remoteproc1: 4a334000.pru0 is available
[   77.251234] pru-rproc 4a334000.pru0: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a334000 probed successfully
[   77.262244] remoteproc remoteproc2: 4a338000.pru1 is available
[   77.262365] pru-rproc 4a338000.pru1: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a338000 probed successfully
[   77.309009] omap-sham 53100000.sham: hw accel on OMAP rev 4.3
[   77.350043] omap-aes 53500000.aes: OMAP AES hw accel rev: 3.2
[   77.355006] omap-aes 53500000.aes: will run requests pump with realtime priority
[   77.933795] remoteproc remoteproc1: powering up 4a334000.pru0
[   77.948829] remoteproc remoteproc1: Booting fw image beaglelogic-pru0-fw, size 62576
[   77.949141] ti-pruss 4a300000.pruss: configured system_events = 0x0000000001d30000 intr_channels = 0x00000037 host_intr = 0x00000037
[   77.949155] remoteproc remoteproc1: remote processor 4a334000.pru0 is now up
[   77.949203] remoteproc remoteproc2: powering up 4a338000.pru1
[   77.986578] remoteproc remoteproc2: Booting fw image beaglelogic-pru1-fw, size 31996
[   77.986674] remoteproc remoteproc2: remote processor 4a338000.pru1 is now up
[   77.986709] BeagleLogic loaded and initializing
[   78.001873] misc beaglelogic: Valid PRU capture context structure found at offset 0000
[   78.001896] misc beaglelogic: BeagleLogic PRU Firmware version: 0.3
[   78.001906] misc beaglelogic: Device supports max 128 vector transfers
[   78.001931] misc beaglelogic: Default sample rate=50000000 Hz, sampleunit=1, triggerflags=0. Buffer in units of 4194304 bytes each
[   79.679691] misc beaglelogic: Successfully allocated 67108864 bytes of memory.
[   80.210149] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   80.210167] Bluetooth: BNEP filters: protocol multicast
[   80.210196] Bluetooth: BNEP socket layer initialized
[   80.364717] Bluetooth: RFCOMM TTY layer initialized
[   80.364767] Bluetooth: RFCOMM socket layer initialized
[   80.364827] Bluetooth: RFCOMM ver 1.11

@abhishek-kakkar If I upload an updated dtb could you please add it to your image and I will see if I can get it pushed up-tree?

sancloud, 2018-05-29:
Updated dtb:
[https://sancloud-my.sharepoint.com/:u:/g/personal/info_sancloud_co_uk/EWQoEwpwCO5Os8NiJ1HyROABZvtDLj1XZKflmnFNWh28dQ?e=VqhAX2](url)

sancloud, 2018-05-29:
@timothy-chapman please copy the file from the link above to:
/boot/dtbs/4.9.36-ti-r46

You can perform that with either inserting your SD card into a linux machine and copying accross (will need sudo) OR scp the file from a Windows PC using FileZilla to the correct location or using a linux PC to scp the file accross.

abhishek-kakkar, 2018-06-04:
@timothy-chapman @sancloud I've updated the image on my side today. Will push the changes when I return today evening and close this issue.

abhishek-kakkar, 2018-06-04:
New image uploaded at https://drive.google.com/uc?export=download&confirm=XgxR&id=1YMSuJthViOqjQhwWWzN_3Pa0IQY5gYKa.

Can it be verified that it works out-of-the-box with SanCloud BeagleBone Enhanced?

closed 2020-07-11 by abhishek-kakkar