abhishek-kakkar/BeagleLogic #32: Add PRUDAQ support to latest BeagleLogic Image

ForceTronics, 2018-07-11:
I recently purchased a BBB and the PRUDAQ cape. When I try to boot up via SD card with the BeagleLogic image that includes the PRUDAQ add-on (image comes from link on PRUDAQ quick start webpage) no network connection appears on my Windows 10 machine and I am not able to SSH into the BBB operating system. When I boot-up from the newest BeagleLogic image  (no PRUDAQ add on) on my BBB a network connection appears on my PC and I am able to log into the BBB using SSH. It seems that the newer BBB boards do not work with the current BeagleLogic image that includes the PRUDAQ add on. Can you add the PRUDAQ code to the newest BeagleLogic image?

abhishek-kakkar, 2018-07-24:
In about 2 weeks I will be releasing a new image for BeagleLogic (and hopefully transition to more regular image releases). I'll get the PRUDAQ support also done then.

If you need this on priority (it's blocking some of your work), then I might be able to support you with a patch until the production image is available.

ForceTronics, 2018-07-24:
Hello Kumar,

Thanks for the follow-up and commitment to update the image with the Prudaq
add on.

We did purchase the Prudaq for a specific project to provide a proof of
concept and to evaluate the ADC. We were hoping that with the example code
we could quickly get to proof of concept stage and then from there leverage
the example code to build our own custom application. If you can provide
some type of patch in the near future that would be greatly appreciated. If
not just having the two options work with the latest BBB hardware and image
would be good.

Please keep us updated,

Neil

rocketBeagle, 2019-10-29:
> In about 2 weeks I will be releasing a new image for BeagleLogic (and hopefully transition to more
> regular image releases). I'll get the PRUDAQ support also done then.
> If you need this on priority [...]

@abhishek-kakkar 
Where do I find the newest beagleLogic-Image?  
The image from the beaglelogic.readthedocs.io is from 2017-07-13.

rocketBeagle, 2020-01-19:
I've looked at the PRU1 assembler code modified for the PRUDAQ board. 
Am I correct in assuming that the beagleLogic samplerate-parameter is actually without function for the PRUDAQ project?

abhishek-kakkar, 2020-07-11:
The release didn't happen as expected - PRUDAQ will need additional work to enable GPIOs and the clock sources for the ADC besides the custom firmware for the PRUs. I'm currently working on a stable image for v5.4 kernels, PRUDAQ support will happen post that.