abhishek-kakkar/BeagleLogic #36: sigrok-cli tcp connection?

xiongyihui, 2018-11-06:
Can sigrok-cli connect to BeagleLogic through TCP?

Tried the following command, but failed.
```
$ sigrok-cli --driver=beaglelogic:conn=192.168.7.2/5555 --scan
sr: hwdriver: Invalid option 'conn'.
```

xiongyihui, 2018-11-06:
Thanks! After recompiled libsigrok, libsigrokdecode and sigrok-cli, it works.

```
sigrok-cli --driver=beaglelogic:conn=tcp_raw/192.168.7.2/5555 -c samplerate=10M --samples 10M -o capture.sr
```

closed 2018-11-06 by xiongyihui

RedFantom, 2018-12-13:
When trying to use the PulseView software and `sigrok-cli` I encounter the same issue. My server is running just fine (I connect to it using `telnet` over the local network, and then my connection was accepted). I have tried compiling and installing `libsigrok`, `sigrok-cli` and `libsigrokdecode` from the latest source available from the sigrok git, but it did not resolve the issue. I can't use PulseView with [this](https://github.com/abhishek-kakkar/libsigrok) version of `libsigrok`, because it is too old. The link for downloading PulseView AppImages in the [documentation](https://standalone.beaglelogic.net/en/latest/getting_started.html?highlight=pulseview) is no longer available. Am I missing something? What sources should I compile and install from? This seems like a really cool project, and I would love to get it to work with PulseView.

skypuppy, 2018-12-13:
Could be the server was down for a bit, Red.  Try again later. Sigrock 
and PulseView have traditionally run neck and neck.  Even if one falls 
behind the other, they should (theoretically) still work as pairs.  
PulseView seems to have wonderful decoders so not sure why sigrock would 
develop them on their own.  I thought they shared decoder code so that 
each could run without the other if it was necessary.

abhishek-kakkar, 2018-12-14:
I have a built AppImage that you can try - https://hackaday.io/project/25745-beaglelogic-standalone/log/69517-documentation-binaries-download-and-instructional-videos .

Please let me know if it works for you.

RedFantom, 2018-12-15:
That `AppImage` works like a charm, thank you!