abhishek-kakkar/BeagleLogic #39: BeagleLogic not working with latest debian image running kernel 4.14

atermos, 2019-01-22:
I am trying to install BeagleLogic on my beaglebone black.
I am using the latest image from beagleboard.org: Debian 9.5 2018-10-07 (with LXQT) http://debian.beagleboard.org/images/bone-debian-9.5-lxqt-armhf-2018-10-07-4gb.img.xz

There does not seem to be an error in the installation but /dev/beaglelogic does not exist after reboot. 
debian@beaglebone:~$ ls -l /dev/beaglelogic
ls: cannot access '/dev/beaglelogic': No such file or directory

I attached my logfile of the outputs of install.sh.
[beaglelogiclog.txt](https://github.com/abhishek-kakkar/BeagleLogic/files/2784408/beaglelogiclog.txt)

I tried troubleshooting using the documentation here: https://beaglelogic.readthedocs.io/en/latest/beaglelogic_system_image.html. 

debian@beaglebone:~$ journalctl | grep beaglelogic 
Jan 22 16:31:37 beaglebone beaglelogic-tcp[1032]: Listening on port 5555
Jan 22 16:31:37 beaglebone bash[1026]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Jan 22 16:33:19 beaglebone beaglelogic[982]: Express server listening on port 4000
Jan 22 16:34:56 beaglebone bash[1026]: beaglelogic-startup: timeout. BeagleLogic couldn't load.
Jan 22 16:34:56 beaglebone systemd[1]: beaglelogic-startup.service: Main process exited, code=exited, status=1/FAILURE
Jan 22 16:34:56 beaglebone systemd[1]: beaglelogic-startup.service: Unit entered failed state.
Jan 22 16:34:56 beaglebone systemd[1]: beaglelogic-startup.service: Failed with result 'exit-code'.

Further running:
debian@beaglebone:~$ cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet

Here is the output of sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT00C03518BBBK0960]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.16.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/beaglelogic-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190117.0-0rcnee0~ stretch+20190119]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~ stretch+20180517]
pkg:[kmod]:[23-2rcnee1~ stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~ stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~ stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai beaglelogic]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.127600] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.139704] gpio-of-helper ocp:cape-universal: ready
END

abhishek-kakkar, 2019-01-23:
You need to downgrade to the 4.9 kernel to use BeagleLogic for now. I am working to bring BeagleLogic kernel module to 4.14 but it will take some time.

sPyOpenSource, 2019-10-20:
@abhishek-kakkar do you need help for that?

rocketBeagle, 2019-10-29:
@abhishek-kakkar How long will it approximately take, till BeagleLogic kernel module is working with 4.14? Thank you very much for doing that work btw!

abhishek-kakkar, 2020-07-04:
Hi, this project had been in limbo for a long period of time, apologies for delayed response.

As of today, BeagleLogic has been successfully ported to the v5.4-ti kernel version. Please wait for an update.

rherban, 2020-07-07:
@abhishek-kakkar 
I work with @atermos and we're willing to help beta-test and validate the new image.  Please let us know how we can help.

abhishek-kakkar, 2020-07-17:
@rherban, there has been some progress; you will hear from me on this thread when I release a new image. Alternatively please reach out on the BeagleLogic Google groups.

mvduin, 2020-12-06:
Currently attempting it to compile for 4.19.94-ti-r42 fails with:
```
/home/debian/BeagleLogic/kernel/beaglelogic.c:25:10: fatal error: linux/pruss_intc.h: No such file or directory
 #include <linux/pruss_intc.h>
          ^~~~~~~~~~~~~~~~~~~~
```
But removing that line allows it to compile successfully.

mvduin, 2020-12-06:
Steps for installing the kernel driver on 4.19 (execute in kernel/ subdirectory of a checkout of this repository, after removing or commenting out the `#include <linux/pruss_intc.h>` line in beaglelogic.c:
- Make sure kernel headers are installed: `sudo apt install linux-headers-$(uname -r)`
- Build the module: `make`
- Install the module: `sudo cp beaglelogic.ko /lib/modules/$(uname -r)`
- Update module index: `sudo depmod`
- Update initramfs: `sudo update-initramfs -u`

skypuppy, 2020-12-06:
What are folks doing on the BBB lately?  I've not picked any of mine up 
in a long time.

David

mvduin, 2020-12-07:
Unfortunately although it compiles and loads fine, it doesn't actually work:
```
beaglelogic pru-beaglelogic: Unable to get PRU0.
```

@skypuppy This is a support ticket, not a chat box. This is a very inappropriate place for asking random off-topic questions.

abhishek-kakkar, 2022-02-20:
As of today the latest kernel version from TI is 5.10 and BeagleLogic will support it (just got it compiling today, need some more testing) - I am trying to set up a CI environment and packaging that should offer ready releases for versions 5.4 and 5.10, unfortunately 4.x versions will be EOL (4.9 will still be there because it's already in the BeagleBoard tree, but no 4.14/4.19 unfortunately).

Closing this ticket because it is referring to 4.14 version which is marked obsolete. Please open a new ticket should you need support with a 5.x version.

closed 2022-02-20 by abhishek-kakkar

mvduin, 2022-02-20:
The latest official beagleboard.org images are still running 4.19, 5.x kernels are only used in testing images.