abhishek-kakkar/BeagleLogic #40: BeagleLogic not working with 4.9.78-ti-r94

majesty-dump, 2019-03-27:
I am trying to install BeagleLogic on my Pocket Beagle.
I am using old image from beagleboard.org: Debian 9.3 2018-01-28 (IOT version) 
### There some logs of commands from troubleshooting:
debian@beaglebone:~$ journalctl | grep beaglelogic
Mar 27 18:29:16 beaglebone systemd[933]: beaglelogic-tcp.service: Failed at step EXEC spawning /home/debian/BeagleLogic/tcp-server-go/server: No such file or directory
Mar 27 18:29:16 beaglebone bash[922]: /home/debian/BeagleLogic/scripts/beaglelogic-startup.sh: line 33: /sys/devices/platform/bone_capemgr/slots: No such file or directory
Mar 27 18:29:16 beaglebone bash[922]: /home/debian/BeagleLogic/scripts/beaglelogic-startup.sh: line 34: /sys/devices/platform/bone_capemgr/slots: No such file or directory
Mar 27 18:29:16 beaglebone bash[922]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Mar 27 18:29:17 beaglebone systemd[1]: beaglelogic-tcp.service: Main process exited, code=exited, status=203/EXEC
Mar 27 18:29:17 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 27 18:29:17 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 27 18:29:17 beaglebone systemd[1]: beaglelogic-tcp.service: Service hold-off time over, scheduling restart.
Mar 27 18:29:19 beaglebone systemd[1038]: beaglelogic-tcp.service: Failed at step EXEC spawning /home/debian/BeagleLogic/tcp-server-go/server: No such file or directory
Mar 27 18:29:19 beaglebone systemd[1]: beaglelogic-tcp.service: Main process exited, code=exited, status=203/EXEC
Mar 27 18:29:19 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 27 18:29:19 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 27 18:29:20 beaglebone systemd[1]: beaglelogic-tcp.service: Service hold-off time over, scheduling restart.
Mar 27 18:29:20 beaglebone systemd[1081]: beaglelogic-tcp.service: Failed at step EXEC spawning /home/debian/BeagleLogic/tcp-server-go/server: No such file or directory
Mar 27 18:29:20 beaglebone systemd[1]: beaglelogic-tcp.service: Main process exited, code=exited, status=203/EXEC
Mar 27 18:29:20 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 27 18:29:20 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 27 18:29:21 beaglebone systemd[1]: beaglelogic-tcp.service: Service hold-off time over, scheduling restart.
Mar 27 18:29:21 beaglebone systemd[1093]: beaglelogic-tcp.service: Failed at step EXEC spawning /home/debian/BeagleLogic/tcp-server-go/server: No such file or directory
Mar 27 18:29:21 beaglebone systemd[1]: beaglelogic-tcp.service: Main process exited, code=exited, status=203/EXEC
Mar 27 18:29:21 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 27 18:29:21 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 27 18:29:21 beaglebone systemd[1]: beaglelogic-tcp.service: Service hold-off time over, scheduling restart.
Mar 27 18:29:21 beaglebone systemd[1105]: beaglelogic-tcp.service: Failed at step EXEC spawning /home/debian/BeagleLogic/tcp-server-go/server: No such file or directory
Mar 27 18:29:22 beaglebone systemd[1]: beaglelogic-tcp.service: Main process exited, code=exited, status=203/EXEC
Mar 27 18:29:22 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 27 18:29:22 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 27 18:29:22 beaglebone systemd[1]: beaglelogic-tcp.service: Service hold-off time over, scheduling restart.
Mar 27 18:29:22 beaglebone systemd[1]: beaglelogic-tcp.service: Start request repeated too quickly.
Mar 27 18:29:22 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 27 18:29:22 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 27 18:29:37 beaglebone beaglelogic[940]: Express server listening on port 4000
Mar 27 18:29:56 beaglebone kernel: beaglelogic pru-beaglelogic: Unable to get pruss handle.
Mar 27 18:29:56 beaglebone kernel: beaglelogic: probe of pru-beaglelogic failed with error -2
Mar 27 18:31:18 beaglebone bash[922]: beaglelogic-startup: timeout. BeagleLogic couldn't load.
Mar 27 18:31:18 beaglebone systemd[1]: beaglelogic-startup.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 18:31:18 beaglebone systemd[1]: beaglelogic-startup.service: Unit entered failed state.
Mar 27 18:31:18 beaglebone systemd[1]: beaglelogic-startup.service: Failed with result 'exit-code'.
### The next one :
 debian@beaglebone:~$ cat /proc/cmdline
console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
### Also there is version.sh :
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335PBGL00A21750EPB00187]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r94]
nodejs:[v6.12.3]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.460734] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END
Is there the same problem as [https://github.com/abhishek-kakkar/BeagleLogic/issues/39](url)

abhishek-kakkar, 2020-07-04:
BeagleLogic doesn't work on the PocketBeagle (yet).

Please use the bigger BeagleBone for the time being, support will be available in a few weeks from now.

abhishek-kakkar, 2022-02-20:
This issue has two components:
(i) TCP support not being available because the Go binary wasn't compiled on the image (so needs a BeagleLogic reinstall)
(ii) The larger issue is that BeagleLogic will in theory run on the PocketBeagle but you will not be able to sample pins as of today because it is sampling PRU1 pins however PocketBeagle has PRU0 lines broken out. Post a PRU swap it may be possible to work. Newer updated BeagleLogic kernel may support this.

Closing this for now, will add an update as and when I can confirm whether BeagleLogic will run on the PocketBeagle or not.

closed 2022-02-20 by abhishek-kakkar