abhishek-kakkar/BeagleLogic #41: BeagleLogic couldn't load

aposstar, 2019-07-03:
Hello everyone,

/proc/cmdline seems logic
```
debian@beaglebone:~$ cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
```
but device node cannot load:
```
debian@beaglebone:~$ journalctl |grep beaglelogic
Jul 03 15:49:54 beaglebone beaglelogic-tcp[516]: Listening on port 5555
Jul 03 15:49:55 beaglebone bash[532]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Jul 03 15:50:21 beaglebone beaglelogic[607]: Express server listening on port 4000
Jul 03 15:51:56 beaglebone bash[532]: beaglelogic-startup: timeout. BeagleLogic couldn't load.
Jul 03 15:51:56 beaglebone systemd[1]: beaglelogic-startup.service: Main process exited, code=exited, status=1/FAILURE
Jul 03 15:51:56 beaglebone systemd[1]: beaglelogic-startup.service: Unit entered failed state.
Jul 03 15:51:56 beaglebone systemd[1]: beaglelogic-startup.service: Failed with result 'exit-code'.
Jul 03 15:52:36 beaglebone sudo[4032]:   debian : TTY=pts/0 ; PWD=/dev ; USER=root ; COMMAND=/usr/sbin/service beaglelogic-startup restart
Jul 03 15:52:36 beaglebone bash[4038]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Jul 03 15:54:38 beaglebone bash[4038]: beaglelogic-startup: timeout. BeagleLogic couldn't load.
Jul 03 15:54:38 beaglebone systemd[1]: beaglelogic-startup.service: Main process exited, code=exited, status=1/FAILURE
Jul 03 15:54:38 beaglebone systemd[1]: beaglelogic-startup.service: Unit entered failed state.
Jul 03 15:54:38 beaglebone systemd[1]: beaglelogic-startup.service: Failed with result 'exit-code'.
Jul 03 16:09:56 beaglebone kernel: bone_capemgr bone_capemgr: part_number 'beaglelogic', version 'N/A'
```
System Image that BBB-wireless is running is : beaglelogic-stretch-2017-07-13-4gb-update.img.xz that seems to solve the same problem for BBE

Any advice would be welcomed

[dmesg_log.txt](https://github.com/abhishek-kakkar/BeagleLogic/files/3355903/dmesg_log.txt)

abhishek-kakkar, 2020-07-04:
Hi, BeagleLogic will shortly be released on the v5.4-ti kernel which should correct the issues that you are seeing.

McNugget6750, 2021-02-27:
Same issue. Just installed on everything latest. Is there any update on this?

McNugget6750, 2021-02-27:
debian@beaglebone:~$ journalctl | grep beaglelogic
Feb 27 04:31:56 beaglebone bash[1104]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Feb 27 04:32:01 beaglebone beaglelogic-tcp[1140]: Listening on port 5555
Feb 27 04:32:44 beaglebone beaglelogic[1113]: Express server listening on port 4000
debian@beaglebone:~$ journalctl | grep beaglelogic
Feb 27 04:31:56 beaglebone bash[1104]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Feb 27 04:32:01 beaglebone beaglelogic-tcp[1140]: Listening on port 5555
Feb 27 04:32:44 beaglebone beaglelogic[1113]: Express server listening on port 4000
Feb 27 04:33:57 beaglebone bash[1104]: beaglelogic-startup: timeout. BeagleLogic couldn't load.
Feb 27 04:33:57 beaglebone systemd[1]: beaglelogic-startup.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 04:33:57 beaglebone systemd[1]: beaglelogic-startup.service: Failed with result 'exit-code'.
debian@beaglebone:~$ cat /
cat: /: Is a directory
debian@beaglebone:~$ proc/cmdline
-bash: proc/cmdline: No such file or directory
debian@beaglebone:~$ cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

McNugget6750, 2021-02-27:
```
debian@beaglebone:~$ journalctl | grep beaglelogic
Feb 27 04:31:56 beaglebone bash[1104]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Feb 27 04:32:01 beaglebone beaglelogic-tcp[1140]: Listening on port 5555
Feb 27 04:32:44 beaglebone beaglelogic[1113]: Express server listening on port 4000
debian@beaglebone:~$ journalctl | grep beaglelogic
Feb 27 04:31:56 beaglebone bash[1104]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Feb 27 04:32:01 beaglebone beaglelogic-tcp[1140]: Listening on port 5555
Feb 27 04:32:44 beaglebone beaglelogic[1113]: Express server listening on port 4000
Feb 27 04:33:57 beaglebone bash[1104]: beaglelogic-startup: timeout. BeagleLogic couldn't load.
Feb 27 04:33:57 beaglebone systemd[1]: beaglelogic-startup.service: Main process exited, code=exited, status=1/FAILURE
Feb 27 04:33:57 beaglebone systemd[1]: beaglelogic-startup.service: Failed with result 'exit-code'.
debian@beaglebone:~$ cat /
cat: /: Is a directory
debian@beaglebone:~$ proc/cmdline
-bash: proc/cmdline: No such file or directory
debian@beaglebone:~$ cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

```

krzyklo, 2021-03-09:
I did update as described here: https://groups.google.com/g/beaglelogic/c/q0NyCB0JOgA

> To update the BeagleLogic side (note: you must be running the 4.9 kernel version and image):
> 
>     cd /opt/BeagleLogic (or whichever folder BeagleLogic has been installed to on the BeagleBone)
>     git pull
>    **sudo** ./install.sh --upgrade

Install golang:
> 1. sudo apt-get install golang
> 2. go run /opt/BeagleLogic/tcp-server-go/server.go

Reboot BBB ( in my case I need to reboot a few times, before it start working). Use the cmds below to confirm all is ok:
`sudo service beaglelogic-tcp status`
`journalctl | grep beaglelogic`
`sudo /opt/scripts/tools/version.sh`

Problems encountered before a few restarts:
```
debian@beaglebone:~$ sudo service beaglelogic-startup status
[sudo] password for debian: 
● beaglelogic-startup.service - BeagleLogic Startup Script
   Loaded: loaded (/lib/systemd/system/beaglelogic-startup.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2021-03-08 19:36:09 UTC; 18s ago
  Process: 832 ExecStart=/bin/bash /opt/BeagleLogic/scripts/beaglelogic-startup.sh (code=exited, status=0/SUCCESS)
 Main PID: 832 (code=exited, status=0/SUCCESS)

Mar 08 19:35:23 beaglebone systemd[1]: Starting BeagleLogic Startup Script...
Mar 08 19:35:24 beaglebone bash[832]: beaglelogic-startup: Waiting for BeagleLogic to show up (timeout in 120 seconds)
Mar 08 19:36:08 beaglebone bash[832]: beaglelogic-startup: Configuring LA pins
Mar 08 19:36:09 beaglebone bash[832]: beaglelogic-startup: Allocating 128MiB of logic buffer to BeagleLogic
Mar 08 19:36:09 beaglebone bash[832]: beaglelogic-startup: Loaded
Mar 08 19:36:09 beaglebone systemd[1]: Started BeagleLogic Startup Script.
debian@beaglebone:~$ sudo service beaglelogic-tcp status
● beaglelogic-tcp.service - BeagleLogic Logic Analyzer TCP Service
   Loaded: loaded (/lib/systemd/system/beaglelogic-tcp.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-03-08 19:35:28 UTC; 2min 14s ago
  Process: 1002 ExecStart=/opt/BeagleLogic/tcp-server-go/server (code=exited, status=203/EXEC)
 Main PID: 1002 (code=exited, status=203/EXEC)

Mar 08 19:35:28 beaglebone systemd[1]: Started BeagleLogic Logic Analyzer TCP Service.
Mar 08 19:35:28 beaglebone systemd[1]: beaglelogic-tcp.service: Main process exited, code=exited, status=203/EXEC
Mar 08 19:35:28 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 08 19:35:28 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 08 19:35:28 beaglebone systemd[1]: beaglelogic-tcp.service: Service hold-off time over, scheduling restart.
Mar 08 19:35:28 beaglebone systemd[1]: Stopped BeagleLogic Logic Analyzer TCP Service.
Mar 08 19:35:28 beaglebone systemd[1]: beaglelogic-tcp.service: Start request repeated too quickly.
Mar 08 19:35:28 beaglebone systemd[1]: Failed to start BeagleLogic Logic Analyzer TCP Service.
Mar 08 19:35:28 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 08 19:35:28 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
debian@beaglebone:~$ sudo service beaglelogic-tcp restart
debian@beaglebone:~$ sudo service beaglelogic-tcp status
● beaglelogic-tcp.service - BeagleLogic Logic Analyzer TCP Service
   Loaded: loaded (/lib/systemd/system/beaglelogic-tcp.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-03-08 19:38:48 UTC; 1s ago
  Process: 2034 ExecStart=/opt/BeagleLogic/tcp-server-go/server (code=exited, status=203/EXEC)
 Main PID: 2034 (code=exited, status=203/EXEC)

Mar 08 19:38:48 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 08 19:38:48 beaglebone systemd[1]: beaglelogic-tcp.service: Service hold-off time over, scheduling restart.
Mar 08 19:38:48 beaglebone systemd[1]: Stopped BeagleLogic Logic Analyzer TCP Service.
Mar 08 19:38:48 beaglebone systemd[1]: beaglelogic-tcp.service: Start request repeated too quickly.
Mar 08 19:38:48 beaglebone systemd[1]: Failed to start BeagleLogic Logic Analyzer TCP Service.
Mar 08 19:38:48 beaglebone systemd[1]: beaglelogic-tcp.service: Unit entered failed state.
Mar 08 19:38:48 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.

```

```
debian@beaglebone:~$ journalctl | grep beaglelogic
[... 28 lines cut ...]
Mar 08 19:46:25 beaglebone systemd[1]: beaglelogic-tcp.service: Failed with result 'exit-code'.
Mar 08 19:46:44 beaglebone beaglelogic[823]: Express server listening on port 4000
Mar 08 19:46:51 beaglebone beaglelogic[823]: Socket.IO connected
Mar 08 19:46:51 beaglebone beaglelogic[823]: Socket.IO connected
Mar 08 19:46:52 beaglebone beaglelogic[823]: Socket.IO connected
Mar 08 19:46:52 beaglebone beaglelogic[823]: Socket.IO connected
Mar 08 19:46:53 beaglebone sudo[1437]:   debian : TTY=pts/0 ; PWD=/home/debian ; USER=root ; COMMAND=/usr/sbin/service beaglelogic-tcp status
Mar 08 19:47:03 beaglebone kernel: remoteproc remoteproc1: Booting fw image beaglelogic-pru0-fw, size 62576
Mar 08 19:47:04 beaglebone kernel: remoteproc remoteproc2: Booting fw image beaglelogic-pru1-fw, size 31996
Mar 08 19:47:04 beaglebone kernel: misc beaglelogic: Valid PRU capture context structure found at offset 0000
Mar 08 19:47:04 beaglebone kernel: misc beaglelogic: BeagleLogic PRU Firmware version: 0.3
Mar 08 19:47:04 beaglebone kernel: misc beaglelogic: Device supports max 128 vector transfers
[... cut ...]
Mar 08 19:49:52 beaglebone beaglelogic[823]: Someone disconnected: "ping timeout"
Mar 08 21:05:15 beaglebone sudo[2037]:   debian : TTY=pts/0 ; PWD=/home/debian ; USER=root ; COMMAND=/usr/sbin/service beaglelogic-tcp status
Mar 08 21:05:20 beaglebone sudo[2043]:   debian : TTY=pts/0 ; PWD=/home/debian ; USER=root ; COMMAND=/usr/sbin/service beaglelogic-tcp status
Mar 08 21:07:25 beaglebone sudo[2083]:   debian : TTY=pts/1 ; PWD=/home/debian ; USER=root ; COMMAND=/usr/sbin/service beaglelogic-tcp status
```

abhishek-kakkar, 2022-02-20:
I suppose that the issue described here (missing TCP binary) has been resolved and hence I am closing the issue. Please reopen if any more help is needed.

closed 2022-02-20 by abhishek-kakkar