abhishek-kakkar/BeagleLogic #48: System image link is broken

xouillet, 2021-11-29:
Hi,

According to the documentation site and the github wiki https://github.com/abhishek-kakkar/BeagleLogic/wiki/BeagleLogic-%22no-setup-required%22-setup:-Introducing-System-Image!, we should be able to download the system image at https://goo.gl/RiXGBs but this link sent me to a 403 error.

mminnella, 2021-12-16:
Same issue, is it available in some other mirrors?

abhishek-kakkar, 2021-12-28:
This issue is because of Google Drive enforcing resource keys on widely shared files. The system images are currently hosted there.

Please use this [updated link](https://drive.google.com/file/d/0ByqRLxWwTOOkSmptSkNqVXBLVm8/view?usp=sharing&resourcekey=0-ZNcpY1XLfVit0cmIbM2-Mw) which has the added resource key string to download.

abhishek-kakkar, 2021-12-28:
Wiki link updated, closing issue.

closed 2021-12-28 by abhishek-kakkar