abhishek-kakkar/BeagleLogic #50: Installing on 5.x

rherban, 2022-04-12:
According to this reply https://github.com/abhishek-kakkar/BeagleLogic/issues/39#issuecomment-1046176815, you are working on a build path for Beaglelogic on a 5.x kernel.  

Do you have any progress here?  Are you going to have a pre-built image available for it?

abhishek-kakkar, 2023-01-18:
Yes, there will be. I plan to set up a GitHub Actions Pipeline to be able to distribute images / Debian repositories to refresh BeagleLogic but it will take a bit more time. Please bear with me while I get this sorted.

rherban, 2023-01-18:
I've tried compiling the beaglelogic firmware on a BeagleBone AI64 but encountered an error (presumably because of a newer clpru).  Can you make your 5.x code available in a branch so we can play around with it?

```
root@BeagleBone:/tmp/BeagleLogic/firmware# make

************************************************************
Building project: firmware

Building file: beaglelogic-pru0.c
Invoking: PRU Compiler
/usr/share/ti/cgt-pru/bin/clpru --include_path=/usr/share/ti/cgt-pru/include --include_path=/usr/lib/ti/pru-software-support-package/include --include_path=/usr/lib/ti/pru-software-support-package/include/am335x -v3 -O2 --display_error_number --endian=little --hardware_mac=on --obj_directory=release --pp_directory=release -ppd -ppa -fe release/beaglelogic-pru0.object beaglelogic-pru0.c
"resource_table_0.h", line 50: error #71: incomplete type is not allowed
"resource_table_0.h", line 88: error #71: incomplete type is not allowed
"resource_table_0.h", line 90: error #28: expression must have a constant value
3 errors detected in the compilation of "beaglelogic-pru0.c".

>> Compilation failure
make: *** [Makefile:105: release/beaglelogic-pru0.object] Error 1
```

abhishek-kakkar, 2023-01-20:
The error that you're seeing is at the PRU firmware compilation level. It appears that the current kernel module source code should compile on the 5.4-ti branch as I had merged it into the main tree.

We might need to update the firmware with the latest TI PRU support package libraries for this to work.

abhishek-kakkar, 2023-01-20:
Also, I haven't tried BeagleLogic on the AI64 yet, since it's a new platform so adding support won't be trivial, you're likely to need work at the pinmux and also device tree level. The SoC has multiple PRUs from what I've heard and firmware should go to the correct one. I've requested for one to try, but let's see when I get one.