abhishek-kakkar/BeagleLogic #53: anyone sell full board?

felixn-unity, 2023-01-01:
Hi

Looks like an amazing project, but is it out commercially?

Thanks

htownclyde, 2023-01-13:
I can't find anyone selling it. I think it has to be built using the KiCAD schematic.

I hope gerbers will be released at some point! Cheers

felixn-unity, 2023-01-13:
damn - but thanks :)

abhishek-kakkar, 2023-01-18:
Unfortunately, BeagleLogic hardware (the standalone version, not the cape) never worked out to be a sustainable board to produce in small quantities - I knew there was demand but wasn't sure if people would pay > 100$ for a device when there are competing high performance single board computers available at that price.

If there is demand for 100 boards this is something I can reconsider, but until then you have to go the DIY way.

The cape design in this repo doesn't have Gerbers, but you can of course build it cheaply with the KiCad sources. You could even use BeagleLogic without the cape, provided you ever only worked with 3.3V signals and never plug anything that could damage your Beagle, the cape just prevents all that.

closed 2023-01-18 by abhishek-kakkar

JarrettR, 2023-01-22:
Hi,

I updated the BOM with real, existing part numbers, and generated manufacturing files so that it's easier for non-EE people to just hit a button and order:
https://www.pcbway.com/project/shareproject/W97133AST112_beaglelogic_c5f4b226.html

You can order just the PCBs, or download the gerbers. I have purchased the PCBs myself and populated them myself, but I make no promises on the PCBA if someone chooses to do so. I've populated the BOM with parts I've used, and I'm confident in it, though.

Let me know if this is not okay, and I'll take it down. I think PCBWay might give me a $0.50 coupon per order, but I dunno, that's not the motivation here.