abhishek-kakkar/BeagleLogic #54: installed on laptop, wasn't supposed to was I?

Randy413, 2023-01-18:
I was reading here; https://beaglelogic.readthedocs.io/en/latest/install.html and following steps. After typing svdo .....install.xx   My laptop terminal told me to reboot. I have not done that yet. Laptop is running Debian 11 Bullseye. But I am starting to comprehend that webpage better now.
  That was supposed to be for the BBB Debian and not my laptop Debian correct? So if I reboot my laptop it is one big Logic analyzer BBB clone correct?
  So, tried svdo apt remove sudo beaglbb (those typed wrong on purpose here, correctly in terminal), It could not find it. Is there a terminal command I can use to reverse the install? Or not boot into that kernel on the laptop boot?
  That page could be a touch clearer maybe, Like 'on the bbb, and not the host laptop that has sigrok' Just an idea is all the above is as I now suspect (if I reboot now)

abhishek-kakkar, 2023-01-18:
Oh my, yes. You're supposed to run the installation on your BeagleBone SSH'ed in, not on your PC.

You can see what the script does and clean up, the key step being to remove the systemd service that tries to autostart BeagleLogic binaries on boot.

Please feel free to open a pull request and fix the docs, it's a part of this repository itself, or I can do that in a bit.

Randy413, 2023-01-18:
Thanks! Very quick reply. I did do some research and I am okay with reinstalling and reload from backup as well. I found that I get an option for booting to advanced or alternate kernels on boot, so presently I may just choose to interrupt boot at that screen and choose the kernel version I want to boot to (new to linux too).
   It is funny, I also always have this and a Windows laptop on the desk anyway so nothing is lost and it's a great learning experinece. Even now with this, with Linux I need not panic. Thanks for this and all the work you do.

closed 2023-01-18 by Randy413