abhishek-kakkar/BeagleLogic #55: beaglelogic image link dead: https://goo.gl/RiXGBs

hanetzer, 2023-03-04:
As the title states.

abhishek-kakkar, 2023-03-06:
I have been receiving individual requests to share the images to people when they ask via the link - file sharing via Google Drive is broken because shared files need a resource key. Please try again and I will approve.

nbgsmk, 2024-06-14:
Hello my friend! 
I wish I knew about your project many years ago. My BeagleBone has been sitting idle in my desk for the last 5-6 years.
Finally I will be able to give it a purpose, and make, for my lab, a much-needed instrument!
Unfortunately, the mentioned system image link is dead and I'm not sure what do you mean by "please try again and I will approve"
Do you need to refresh the shared file on google drive somehow? I need to contact you directly in some way, so you share the link directly to me? Please let me know.

If not possible, I am completely willing to build the image myself. But the only instructions I could find are here: https://github.com/PhischDotOrg/BeagleLogic-CookBook?tab=readme-ov-file
I wonder if that is worth giving a try?
In particular, there is mention of cloning https://github.com/PhischDotOrg/BeagleLogic.git and building from there.
Today its' 2024, and the latest Debian image is different than mentioned there, but anyway, is this the right path to take?

If not, then please advise on what to do.

MUCH appreciated in advance!
Thanks!
Zoran